MapsHD/HDMapping #25: Dataset?

antun008, 2024-03-09:
Can you give us dataset and guide for saving las, imu and gsss into usb ?

antun008, 2024-03-09:
what exactly do i need to do to get las, imu and gnss files. Should I follow the instructions you wrote in Mandeye? Wiring manual, and other things? For me this is to difficult.

Can you explain this part better, please. Beacsue i downloaded this exe file from hdmapping, but dont know how to get datasets and I also having Livox mid 360 and working with fastlio

JanuszBedkowski, 2024-03-09:
Please provide me Your email so I can send some data

antun008, 2024-03-09:
Thanks a lot!:))
antun006@gmail.com

closed 2024-04-22 by JanuszBedkowski

CharlieV5, 2024-05-27:
> Please provide me Your email so I can send some data

Hi!  I would like to have some example data too ! Here is my e-mail: charliev5@qq.com
Thank you very much!