MapsHD/HDMapping #37: getting stuck in step 1

yuckinus, 2024-06-03:
Hello

Was in balkan cavers camp and we tried the collected data on your machine which processed fine for steps 1 and 2. Got back home and trying to do the same but getting stuck in step 1, tried two times

[image]

antun008, 2024-06-03:
If I'm right:
If the window in the program does not turn white when you click on it, it indicates that the program has created a process and you can proceed to the next step in step 1. Honestly, I don't know why it doesn't show in the last line of code that it has reached the end, but in reality, it seems to me that it practically goes through everything. Conclusion: just try to continue processing as if everything is fine.

JanuszBedkowski, 2024-06-03:
I add 
std::cout << "step_1_done please click 'compute_all (step 2)' to continue calculations" << std::endl; 

[image]

closed 2024-06-03 by JanuszBedkowski