MapsHD/HDMapping #39: HDmapping step 3 processing from the perspective of geodetic surveyor

antun008, 2024-06-04:
Hello to everyone,

I would like to show a video located at the link below this text, which illustrates my perspective on georeferencing a point cloud obtained using HD mapping. On the first day, I flew over the same area with a drone, and on the second day, I captured the same area with Mandeye. Regarding the drone data, I also used a GPS antenna to measure GCPs (one GCP every 70 meters) to precisely position the drone data on our official coordinate system (in the video, this is the RGB point cloud). After obtaining the georeferenced point cloud from the drone, I aimed to align it with the point cloud obtained from Mandeye and HD mapping to get georeferenced pointcloud from lidar.

LINK:
https://youtu.be/9kN_w5f3Vhka

Best regards,
Antun Jakopec

JanuszBedkowski, 2024-08-30:
Please check if https://github.com/MapsHD/HDMapping/releases/tag/v0.57 is solving this issue.

JanuszBedkowski, 2024-08-31:
solved

closed 2024-08-31 by JanuszBedkowski