MapsHD/HDMapping #41: Windows 10 - calculations turning off due to too little RAM - (partially resolved)

MacFryn, 2024-06-06:
Dynamic virtual memory allocation does not work properly in Windows 10. Despite the dynamic memory allocation setting, attempted calculations in step 1 or later may be interrupted due to lack of memory. In this case, the application simply closes without any warning.

**Optional:** it would be good to handle such an error by adding a warning to the user that he has run out of memory.

**Necessary:** ​​Add instructions in the manual on how to deal with this problem.

**To check:**
What is the maximum relative volume of calculation files for the heaviest calculations. Initially for step 1 I think it's about 25x the weight of the input files. I suggest (I guess) using 50x file size as a safe value.

**Change procedure for windows 10:**

1. Prepare enough free space on hard drive
2. Open Settings.
3. Click on System.
4. Click on About.
5. Click the "Advanced system settings" option inside the "Related settings" section.
6. Click the Advanced tab.
7. Click the Settings button inside the "Performance" section.
8. Click the Advanced tab.
9. Click the Change button inside the "Virtual memory" section.
10. Turn off the "Automatically manage paging files size for all drives" option.
11. Select the Custom size option.
12. Select the fastest drive (fast SSD if possible) in „Paging file size for each drive”
13. Specify the initial and maximum size for the paging file in megabytes – Initial 30x files size, maximum – 50x files size.
14. Click the Set button. 
15. Click the OK button and OK button again.
16. Restart your computer

To **rever changes**, just check the „Automatically manage paging files size for all drives" option on step 8, click OK and restart your computer.

michalpelka, 2024-06-11:
Hello, thanks for the detailed manual on Windows 10.
I think it would be beneficial to catch that the user has the environment not configured optimally. 
Next, we should gracefully end the process, or cancel the operation, with a link to some kind of knowledge base. 
In this knowledge base, we should educate users of all supported systems (Windows 10, Windows 11 and Linux) on setting up paging/swap.

CharlieV5, 2024-06-12:
@michalpelka @JanuszBedkowski  Why are the requirements for RAM and CPU performance so high? I tested the first data of OmniWarsawDataset (https://github.com/MapsHD/OmniWarsawDataset), 431 laz files and 431 imu files(2.6GB occupied disk space), which require more than 40GB memory .

MacFryn, 2024-06-12:
> I will copy your detailed tutorial @MacFryn and mark you as a co-author when I create a place for that. Please confirm if you are fine with that @MacFryn.

I posted this post to support the project. feel free to use it.

I just want to point out that the virtual memory values ​​I suggested are overestimated. The point is that users usually have other active applications apart from the operating system. In an extreme case, e.g. a Windows 10 with 4GB RAM, everything is done on virtual memory. Then this high value of virtual memory may actually be needed.

JanuszBedkowski, 2024-06-13:
> requirements for RAM performance so high?

All necessary data for calculations are stored in RAM for better performance. For large scale projects it is recommended to use Win11 with available storage to cope with this issue. I am not sure if refactoring/optimizing code will provide satisfactory results since it is much easier to find Win11 machine with 64GB ram.

closed 2024-08-04 by JanuszBedkowski