MapsHD/HDMapping #62: Other Livox Lidar support

85256638, 2024-08-13:
Thanks for sharing this outstanding work!
I was wondering whether this project can be modified to support other non-repetitive lidars from Livox such as the Avia and Horizon?
If so, what changes i need to adapt?

Thanks!

JanuszBedkowski, 2024-08-13:
It is working out of the box with Livox hap https://github.com/JanuszBedkowski/mandeye_controller.

Livox AVIA requires LIVOX SDK1, thus little effort is needed to prepare data.

Generally speaking, it should work for LiDAR with synchronized IMU.

[image]

85256638, 2024-08-14:
Thanks a lot, I shall modify the driver to replace it with the Livox SDK1.
In terms of the accuracy, how does this work compare with FASTLIO2?
Thanks again!

JanuszBedkowski, 2024-08-14:
Before it please check https://github.com/JanuszBedkowski/mandeye_controller/tree/mp/livox_sdk1_vn100

This was tested with AVIA 

The comparison with FASTLIO2 is in progress.
Please keep in mind that FASTLIO2 is for mobile robots, software here is offline (dedicated for Geo-Science) thus it is not so good idea to compare it.

closed 2024-08-19 by JanuszBedkowski