MapsHD/HDMapping #70: Step 3 - export many clouds as one

DariuszWilczewski, 2024-10-03:
Hi,

I need to export two point clouds, measured on different days, as one cloud. Is there any trick to do this in Step 3? Currently, I have to use Step 2, export the two clouds, and process them as two separate clouds.

I need to rasterize this project using CloudCompare, but it must be a single cloud to do it

JanuszBedkowski, 2024-10-04:
You can concatenate these point clouds in step 2 and export as single point cloud also in step 2.

1. Create new folder and copy ther laz files.
2. Uncheck 'simple_gui'
3. Uncheck 'decimate during load'
4. Load 2 point clouds by using button 'load AlignedPointCloud from WHU-TLS ...'
5. Sava single cloud using button 'save all marked scans to laz (as one global scan)'

Please let me know if it is solving Your issue?

closed 2024-10-05 by JanuszBedkowski

DariuszWilczewski, 2024-10-07:
Sorry for delay.

Yes, is it is a solved isue