MapsHD/HDMapping #71: Consistency vs Consistency2

bo-rc, 2024-10-09:
Hi there are one `Consistency` and the other `Consistency2` buttons in the GUI app `lidar_odometry_step_1`. 

Are they providing the same effects? I notice that `Consistency2` is much slower. I'm wondering if there is a quality / speed trade-off between the two?

[image]

antun008, 2024-10-17:
I think consistency2 is better for indoor

JanuszBedkowski, 2024-10-17:
[image]

I updated GUI

JanuszBedkowski, 2024-10-17:
I hope GUI explains everything

closed 2024-10-17 by JanuszBedkowski