MapsHD/HDMapping #77: Question regarding processing stopscan dataset

MattiasTancred, 2024-12-10:
Hi, big thanks for all the work put into this repo and the accompanying controller repo.

I have tested processing a stopscan dataset for the first time.

It might be due to my limited knowledge, but I am doing something similar to continuous scanning. Then I load session.json in step 2 and get the following.
[image]

GeomPozzoli, 2024-12-12:
Hi Mattias,
I think scans done in stop scan mode should be managed like scans done with a static laser scanner, therefore individual laz files can be registered with other software such as cloudcompare (very complicated), I have never tried to manage them as if they were mobile, I will try to do some experiments
Stefano

JanuszBedkowski, 2024-12-12:
Hi, step 2 multi_view_tls_registration_step_2.exe can do this 
[image]

[image]

all methods are elaborated here 
https://www.sciencedirect.com/science/article/abs/pii/S0263224123007637

JanuszBedkowski, 2024-12-12:
Please use gizmo to move scans

[image]

JanuszBedkowski, 2024-12-12:
use e.g. NDT for multi view registration

[image]

JanuszBedkowski, 2024-12-12:
I think I lost documentation for this :), not so many people were interested in this particular functionality.
Software supports ICP, NDT, point to point, point to plane, plane to plane, point to projection, different angle reprezentation such as Tait Bryan, quaternion, Rodrigues ...

JanuszBedkowski, 2024-12-20:
Ok

closed 2024-12-20 by JanuszBedkowski