MapsHD/HDMapping #84: How to registrate multi-view point clouds

CharlieV5, 2025-02-14:
Hi, dear author!
I have some multi-view Terrestrial Laser Scanner point clouds in xyz format and the pose(position and rotation) of each TLS station. I'd like to test the function of multi_view_tls_registration_step_2.exe but don't know how to input my data , since I don't have the session.json file and other csv files, according to the result of quick_start_demo. 
Any method to transfer my data into the data for multi_view_tls_registration_step_2.exe ?
Thank you!

[test-data.zip](https://github.com/user-attachments/files/18792895/test-data.zip)

JanuszBedkowski, 2025-02-14:
Can You prepare data in such format?

https://3d.bk.tudelft.nl/liangliang/publications/2019/plade/resso.html

Once You have matrices in RESSO.reg

the You change pointslouds into LAZ files.

Use following functions

1. loadAlignedPointCloud from WHU-TLS ....
2. update initial poses from RESSO file
3. update poses from RESOO file

[image]

then play with
1. Normal Distributions transform
2. Iterative Closest Point
3. Plane Features
....

CharlieV5, 2025-02-14:
@JanuszBedkowski Thank you! I will give it a try.

closed 2025-04-21 by JanuszBedkowski