MapsHD/HDMapping #106: Include a "<mutex> into lidar_odometry_step_1/lidar_odometry.cpp

mag72, 2025-03-17:
I have an issue when compiling the latest version from git:
```
HDMapping/apps/lidar_odometry_step_1/lidar_odometry.cpp: In function ‘bool load_data(std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&, LidarOdometryParams&, std::vector<std::vector<Point3Di> >&, Imu&)’:
HDMapping/apps/lidar_odometry_step_1/lidar_odometry.cpp:157:14: error: ‘mutex’ is not a member of ‘std’
  157 |         std::mutex mtx;
      |              ^~~~~
HDMapping/apps/lidar_odometry_step_1/lidar_odometry.cpp:2:1: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
    1 | #include "lidar_odometry.h"
  +++ |+#include <mutex>
    2 | 
HDMapping/apps/lidar_odometry_step_1/lidar_odometry.cpp: In lambda function:
HDMapping/apps/lidar_odometry_step_1/lidar_odometry.cpp:189:22: error: ‘unique_lock’ is not a member of ‘std’
  189 |                 std::unique_lock lck(mtx);
      |                      ^~~~~~~~~~~
HDMapping/apps/lidar_odometry_step_1/lidar_odometry.cpp:189:22: note: ‘std::unique_lock’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
```

Solved by including the mentioned `<mutex>` header.

JanuszBedkowski, 2025-03-17:
What OS and other build details?

mag72, 2025-03-18:
Ubuntu 20.04 / GCC 10.
Just followed this instruction - https://github.com/MapsHD/HDMapping/tree/main#building-commands

marcinmatecki, 2025-03-18:
On Ubuntu 22.04/Windows 10, it compiles without any issues. The problem might be an old version of the C++ compiler on Ubuntu 20.04. Could you please create a PR with the added #include <mutex>? We will check if it affects the functionality of the application, and it might be a solution for older systems. Thank you.

JanuszBedkowski, 2025-04-21:
done

closed 2025-04-21 by JanuszBedkowski