MapsHD/HDMapping #171: Convert ros2 bags to mandeye format

dbaldzhiev, 2025-09-07:
I am trying to convert my ros2 bags to mandeye format but having trouble implementing the maneye format and getting the whole thing to write files that can be properly iterpreted by HDMapping.

have a look here 
https://github.com/dbaldzhiev/bag2mandeye

Regards,
D

JanuszBedkowski, 2025-09-07:
can You check if this is working?
https://github.com/MapsHD/mandeye_to_bag/tree/main/mandeye_to_rosbag2

we provide ROS/ROS2 converters here
https://github.com/MapsHD/mandeye_to_bag

JanuszBedkowski, 2025-09-13:
> As stated in the description: "This repository contains a simple tool that
> enables the conversion of Mandeye data to ROSBag."
>
> Is it somehow two way conversion? I have a *bag* that i want to convert
> to a mandeye files.

You can find both conversions

mandeye_to_bag/mandeye_to_rosbag1/rosbag1_to_mandeye_node.cpp at main ·
MapsHD/mandeye_to_bag
<https://github.com/MapsHD/mandeye_to_bag/blob/main/mandeye_to_rosbag1/rosbag1_to_mandeye_node.cpp>

mandeye_to_bag/mandeye_to_rosbag2/rosbag2_to_mandeye_node.cpp at main ·
MapsHD/mandeye_to_bag
<https://github.com/MapsHD/mandeye_to_bag/blob/main/mandeye_to_rosbag2/rosbag2_to_mandeye_node.cpp>

jb

JanuszBedkowski, 2025-09-13:
bag --> mandeye

https://github.com/MapsHD/mandeye_to_bag/blob/main/mandeye_to_rosbag1/rosbag1_to_mandeye_node.cpp
https://github.com/MapsHD/mandeye_to_bag/blob/main/mandeye_to_rosbag2/rosbag2_to_mandeye_node.cpp